lichtblick-suite/lichtblick #153: Cannot connect through Foxglove bridge

mvanlobensels, 2024-08-26:
Dear Lichtblick team,

Lichtblick worked perfectly for me when connecting to my ROS2 system through the Foxglove bridge. Recently, this stopped working out of the blue. I have the feeling this happened when I updated my system, which upgraded `ros-humble-foxglove-bridge` to version 0.8.0. Have you experienced this issue?

- Version: 1.3.6
- Platform: Pop_OS 22.04
- Data source: ROS2 native through Foxglove bridge

luluiz, 2024-09-11:
Hi @mvanlobensels, 
I've opened an image and 3D panel using a WS connection and it works properly.
Could you check once again and provide more information about this behaviour?
If it persists, I'd say to run in dev mode to monitor the logs via the "Console" tab and provide us with more information.

luluiz, 2024-09-16:
I will close the issue due to lack of response from the user, but if it occurs again, please feel free to reopen the issue.

closed 2024-09-16 by luluiz