lichtblick-suite/lichtblick #227: Text missing on MacOS Menubar

KisonHe, 2024-10-18:
**Description**

Some text missing on MacOS menu bar.

- Version: 1.5.0
- Platform: MacOS Sequoia 15.0.1, M3 Chip
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Not related

**Steps To Reproduce**

1. Open the app, and click menubar. 

[
[image]
](url)

**Expected Behavior**

Texts should be there

laisspportugal, 2024-10-21:
Hi @KisonHe, thank you for your feedback.
We are investigating this issue on our side.

Woodii1998, 2024-10-28:
> Hi @KisonHe, thank you for your feedback. We are investigating this issue on our side.

Do you need help, I can help fix this problem

aneuwald-ctw, 2024-11-04:
@KisonHe we are closing this issue, as it was resolved in the PR #237 and is also available in the new release of version 1.7.0.
https://github.com/Lichtblick-Suite/lichtblick/releases/tag/v1.7.0

Thanks for your contribution and hope you keep enjoying and contribution to Lichtblick!

closed 2024-11-04 by aneuwald-ctw