lichtblick-suite/lichtblick #260: Lichtblick does not connect ROS2 bridge

SvenMuc, 2024-11-14:
**Description**
Lichtblick does not connect to ROS2 bridge on systems with ROS2 Jazzy.

- Version: 1.7.0
- Platform: Ubuntu 24.04.1 LTS + ROS Jazzy
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS Jazzy and mcap replay

**Steps To Reproduce**

1. Launch ROS2 bridge `ros2 launch foxglove_bridge foxglove_bridge_launch.xml`
2. Run Lichtblick tool and connect via ROS2 bridge (ws://localhost:8765)
 
[image]

**Ros2 Bridge Log**
```
ros2 launch foxglove_bridge foxglove_bridge_launch.xml
[INFO] [launch]: All log files can be found below /home/XXX/.ros/log/2024-11-14-09-46-38-656236-D2209234-59676
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [foxglove_bridge-1]: process started with pid [59680]
[foxglove_bridge-1] [WARN] [1731573998.785070978] [rcl]: ROS_LOCALHOST_ONLY is deprecated but still honored if it is enabled. Use ROS_AUTOMATIC_DISCOVERY_RANGE and ROS_STATIC_PEERS instead.
[foxglove_bridge-1] [WARN] [1731573998.785082788] [rcl]: 'localhost_only' is enabled, 'automatic_discovery_range' and 'static_peers' will be ignored.
[foxglove_bridge-1] [INFO] [1731573998.798865835] [foxglove_bridge_component_manager]: Load Library: /opt/ros/jazzy/lib/libfoxglove_bridge_component.so
[foxglove_bridge-1] [INFO] [1731573998.804425503] [foxglove_bridge_component_manager]: Found class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
[foxglove_bridge-1] [INFO] [1731573998.804441447] [foxglove_bridge_component_manager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
[foxglove_bridge-1] [INFO] [1731573998.806279325] [foxglove_bridge]: Starting foxglove_bridge (jazzy, 0.8.0@) with WebSocket++/0.8.2
[foxglove_bridge-1] [INFO] [1731573998.806550425] [foxglove_bridge]: [WS] Server running without TLS
[foxglove_bridge-1] [INFO] [1731573998.806702474] [foxglove_bridge]: [WS] WebSocket server listening at ws://0.0.0.0:8765
[foxglove_bridge-1] [INFO] [1731573998.806706530] [foxglove_bridge]: [WS] WebSocket server run loop started
[foxglove_bridge-1] [INFO] [1731574024.941844691] [foxglove_bridge]: [WS] Rejecting client 127.0.0.1:43646 which did not declare support for subprotocol foxglove.websocket.v1
[foxglove_bridge-1] [ERROR] [1731574024.941935779] [foxglove_bridge]: [WS] Handshake ended with HTTP error: 400
[foxglove_bridge-1] [INFO] [1731574028.125915841] [foxglove_bridge]: [WS] Rejecting client 127.0.0.1:43652 which did not declare support for subprotocol foxglove.websocket.v1
[foxglove_bridge-1] [ERROR] [1731574028.126055366] [foxglove_bridge]: [WS] Handshake ended with HTTP error: 400
[foxglove_bridge-1] [INFO] [1731574031.132082523] [foxglove_bridge]: [WS] Rejecting client 127.0.0.1:42924 which did not declare support for subprotocol foxglove.websocket.v1
[foxglove_bridge-1] [ERROR] [1731574031.132167467] [foxglove_bridge]: [WS] Handshake ended with HTTP error: 400
[foxglove_bridge-1] [INFO] [1731574034.137387822] [foxglove_bridge]: [WS] Rejecting client 127.0.0.1:42938 which did not declare support for subprotocol foxglove.websocket.v1
[foxglove_bridge-1] [ERROR] [1731574034.137493187] [foxglove_bridge]: [WS] Handshake ended with HTTP error: 400
```

**Expected Behavior**
Lichtblick connects to ros2 bridge correctly. On machines with ROS2 Humble everything works as expected.

laisspportugal, 2024-11-14:
Hello @SvenMuc, thank you for your feedback and report. 
We are going to investigate it internally and let you know as soon as we have more information about it.

Best regards, 
Lichtblick Team

tik0, 2025-01-13:
@SvenMuc have you tried the updated version [1.8.0](https://github.com/Lichtblick-Suite/lichtblick/releases/tag/v1.8.0)?

aneuwald-ctw, 2025-02-05:
Hey @SvenMuc, sorry about the delay!

We tried to replicate your issue on our latest release, [v1.9.0](https://github.com/Lichtblick-Suite/lichtblick/releases/tag/v1.9.0), using the same commands you provided. Everything worked as expected. My testing was done on macOS using Docker, and we also verified it on WSL without any issues. To ensure consistency, we followed this Foxglove guide for ROS 2 setup: [Installing ROS 2 on macOS with Docker](https://foxglove.dev/blog/installing-ros2-on-macos-with-docker).

[image]

Additionally, we tested on [v1.7.0](https://github.com/Lichtblick-Suite/lichtblick/releases/tag/v1.7.0), and it also worked as expected.

Based on the error message, it seems Lichtblick may not be able to reach your ROS 2 bridge. Please double-check that both are running in the same network.

Since it's been a while since your last message and we couldn’t reproduce the issue, we will go ahead and close this for now. However, feel free to reopen it or reach out if you need further assistance.

Thanks for using Lichtblick! 🚀

closed 2025-02-05 by aneuwald-ctw

aneuwald-ctw, 2025-02-05:
Additionally, we tested to run a native ROS 2 bridge, using the command:

`ros2 launch rosbridge_server rosbridge_websocket_launch.xml`

That way, it creates a proper ROS 2 bridge using the port 9090, and than you can successfully connect to Lichtblick.

[image]

SvenMuc, 2025-02-06:
Thank you for your support. 
I tested v1.9.0 and it works again. Both the native ROS2 bridge as well as the floxglove bridge.