lichtblick-suite/lichtblick #262: Error: corepack: command not found

ClemensLinnhoff, 2024-11-15:
**Description**

- Version: current main
- Platform: Ubuntu 24.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): not relevant as problem during install

Following the install instructions in the Readme I get the error `corepack: command not found`.
Is there maybe a dependency not listed? The Readme only states Node.js and Git LFS, which I have both installed.

**Steps To Reproduce**
- Install node.js with `apt install nodejs`
- Verify version: v18.19.1 > v16.10
- Run `corepack enable`

**Expected Behavior**
No idea, I am just following the Readme. But I guess there should not be an error.

ctw-joao-luis, 2024-11-15:
Hey @ClemensLinnhoff it's possible that by installing Node.js via apt that corepack could be missing, Try these steps to see if this fixes it:

1. Set up the NodeSource repository on your Ubuntu system to install a specific version of Node.js 
`curl -fsSL https://deb.nodesource.com/setup_18.x | sudo -E bash -
`

2. Install Node.js and npm:
`sudo apt install -y nodejs
`

3. Verify if Node and corepack are actually installed via:
`node -v 
corepack --version`

Best regards, 
Lichtblick Team

ClemensLinnhoff, 2024-11-18:
Unfortunately, this also did not work. I made sure to purge the installed nodejs before. But during the 2. step I get the following error:

```
dpkg: error processing archive /var/cache/apt/archives/nodejs_18.20.4-1nodesource1_amd64.deb (--unpack):
 trying to overwrite '/usr/include/node/common.gypi', which is also in package libnode-dev 12.22.9~dfsg-1ubuntu3.6
```

I could however install the prebuilt debian package of release 1.7. So I will just use that instead of building Lichtblick from source.