lichtblick-suite/lichtblick #267: Camera synchronization failure in the 3D panel.

ToyYan, 2024-11-15:
**Description**
Camera synchronization failure in the 3D panel.

https://github.com/user-attachments/assets/73c64d0b-b16f-4046-a18f-4459b05e2f8b

- Version: 1.7.0
- Platform: macos
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

**Expected Behavior**

ctw-joao-luis, 2024-11-16:
Hi @ToyYan, thanks for reporting this issue! I've identified the root cause of the bug and am currently working on final details to ensure the fix is stable. You can check the fix here #263 

Best regards, 
Lichtblick Team

ctw-joao-luis, 2024-11-20:
Hey again @ToyYan, it's just to inform you that the PR was merged and the fix is live on the main branch, the next release will no longer have issues with camera sync!
If you have any question or more bugs to report don't be afraid to reach out!

Best regards,
Lichtblick Team

closed 2024-11-20 by ctw-joao-luis