lichtblick-suite/lichtblick #277: template extension does not work

medivhc, 2024-11-22:
**Description**
I create a simple extension like https://github.com/Lichtblick-Suite/create-lichtblick-extension/blob/main/template/src/ExamplePanel.tsx.
The topics doesn't show in the panel, while i works in foxglove.

- Version:
- Platform:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

Create a template extension, 
npm init foxglove-extension@latest helloworld

build it and dnd it to lichtblick. 
no topic in the list 
[image]

**Expected Behavior**
should show the topic

ctw-joao-luis, 2024-11-22:
Hi @medivhc,

We’re going to investigate this issue internally to identify what’s going wrong. Could you please let us know which version of Foxglove it works with?

Best Regards,
Lichtblick Team

medivhc, 2024-11-25:
Hi,

I was able to reproduce this issue with Lichtblick version 1.7.19. I am using the web version of Lichtblick. Foxglove.dev works fine.

Basically, context.onRender does not trigger after the topics change？

laisspportugal, 2024-12-06:
Hi @medivhc 
We realized that the problem was related to the React library, that was fixed on Foxglove project but we were lacking on sync. 
The method used before was deprecated and needed to be updated. Since now we updated the project if you create a new library the problem will not gonna happen again. 

Thank you for your feedback!

laisspportugal, 2024-12-06:
We will close