lichtblick-suite/lichtblick #307: getting error while launching desktop Application

AnkitSinghR, 2024-12-14:
**Description**

- Version:"1.8.7",
- Platform: ubuntu 22
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

$ yarn desktop:serve  
$ yarn desktop:start
lichtblick is loading but blank screen is showing
attaching screenshot for your reference

**Expected Behavior**

laisspportugal, 2024-12-16:
Hi @AnkitSinghR

Can you please try to start the application enabling the  LIGBL_ALWAYS_SOFTWARE=1 flag? 
Please let us know the result that you got doing that.
Thanks in advance.

go-sakayori, 2024-12-16:
@laisspportugal 
Hi, I'm having the same problem with version "1.8.5". 
Just wanted to share that I tried 
```$ LIBGL_ALWAYS_SOFTWARE=1 yarn desktop:start```,
however the same error occurred.
I'm trying to solve it, but do you have any idea where to start from?

AnkitSinghR, 2024-12-16:
[image]
Hi @laisspportugal 
still have the same issue
I have  run the commnad which you have mentioned
$ LIBGL_ALWAYS_SOFTWARE=1 yarn desktop:start
[image]

aneuwald-ctw, 2024-12-16:
Hey @AnkitSinghR  thank you for the report, we've identified where the problem is and we're going to work on it.

Best regards.

aneuwald-ctw, 2024-12-16:
@go-sakayori and @AnkitSinghR the issue should be solved with the merge of #308.

Thanks for spotting it and don't hesitate to reach out again if needed!

closed 2024-12-16 by aneuwald-ctw

go-sakayori, 2024-12-16:
@aneuwald-ctw Thank you for your quick response. 
I've tried with the latest source, but the same error occurred. 
Here's a screenshot of what happend.
[image]

I tried
- ```rm -rf node_modules```, then ```yarn desktop:serve``` & ```yarn desktop:start```
- ```yarn run clean```, then ```yarn desktop:serve``` & ```yarn desktop:start```
- remove the local git repository and clone -> yarn install

Any other things I should try?
Also should I create a new issue, since this one is closed?

go-sakayori, 2024-12-18:
@AnkitSinghR 
Just wondering if your problem was solved. Have you tried with the newest version?