lichtblick-suite/lichtblick #313: Desktop app cannot be launched due to Chrome extensions

go-sakayori, 2024-12-20:
**Description**
[image]

- Version: 1.8.11
- Platform: Ubuntu 22.04.4 LTS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): NA

**Steps To Reproduce**
Cloning the main branch then,
```
$ corepack enable
$ yarn install
$ yarn desktop:serve       
$ yarn desktop:start or LIBGL_ALWAYS_SOFTWARE=1 yarn desktop:start
```
The error msg is 
```
Installing Chrome extensions for development...
(node:233072) UnhandledPromiseRejectionWarning: Error: Invalid header: Does not start with Cr24
    at crxToZip (webpack-internal:///../../node_modules/unzip-crx-3/dist/index.js:32:15)
    at eval (webpack-internal:///../../node_modules/unzip-crx-3/dist/index.js:69:32)
    at callHanler (webpack-internal:///../../node_modules/yaku/lib/yaku.js:719:16)
    at tryCatcher (webpack-internal:///../../node_modules/yaku/lib/yaku.js:436:32)
    at eval (webpack-internal:///../../node_modules/yaku/lib/yaku.js:575:38)
    at flush (webpack-internal:///../../node_modules/yaku/lib/yaku.js:478:17)
    at process.processTicksAndRejections (node:internal/process/task_queues:77:11)
(Use `electron --trace-warnings ...` to show where the warning was created)
(node:233072) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
```

**Expected Behavior**
GUI could be launched when launching electron.

**Remarks**
This is the same issue as https://github.com/Lichtblick-Suite/lichtblick/issues/307 which is supposed to be fixed in https://github.com/Lichtblick-Suite/lichtblick/pull/308. The source code works on another desktop with the same version of yarn.
(Was not able to reopen the issue, so made a new issue instead)

- node -v: v22.12.0
- npm -v : 10.9.0
- yarn -v: 1.22.22

``` $yarn run web:serve``` works perfectly, but would like to use my local extension on the desktop app.
Will be happy to provide more data if needed.

aneuwald-ctw, 2024-12-20:
Hey @go-sakayori thanks for reaching out again!

So, you mentioned that in another desktop it works. Maybe it is something related to the cache of your yarn?

Try those:
`yarn clean`
`yarn cache clean`
`rm -rf node_modules`
`yarn install  --check-files`

Because in the first issue we were able to replicate the error, but now our team can't. Could you confirm you tried those commands?

go-sakayori, 2024-12-23:
@aneuwald-ctw Sorry for my late reply.
I agree it's something related to yarn.
I've tried the commands you mentioned, and I had a error as belwo:
```
Unknown Syntax Error: Unsupported option name ("--check-files").
```

What I've done is totally remove the git repository that I've cloned, and re-cloned with a new one. Same issues occurs, so I've cleaned the cache and removed the node_modules (every command worked).  Then installed packages by
``` yarn install`` since ``` yarn install --check-files`` wouldn't work.

go-sakayori, 2025-01-15:
@aneuwald-ctw  I would like to update my situation. 
The problem seems to be solved by adding  `"electron-devtools-installer": "4.0.0"` to `"devDependencies"` in package.json. (Version 3.2.0 worked as well as 4.0.0).
I am not really familiar with typescript (nor javascript), so could you let me know if this solution is appropriate? If there isn't any problem I recommend to add this package.

closed 2025-01-22 by laisspportugal

go-sakayori, 2025-01-22:
@laisspportugal Thank you for fixing the issue.
It is now working with my machine!