lichtblick-suite/lichtblick #323: Failed to build Docker image on latest main

t0ny-peng, 2025-01-13:
**Description**

I'm trying to build the web app using the commit 02d9c40e54f2458761773b53613fe0f3050a02bf. However, the Docker command failed with the following message. 

```
[DESKTOP]➜ /home/name/git/lichtblick git:(main) ✗ docker build . -t lichtblick
[+] Building 21.8s (13/16)                                                                                                  
 => [internal] load .dockerignore                                                                                                                                                                                                                  0.0s
 => => transferring context: 111B                                                                                                                                                                                                                  0.0s
 => [internal] load build definition from Dockerfile                                                                                                                                                                                               0.0s
 => => transferring dockerfile: 801B                                                                                                                                                                                                               0.0s
 => [internal] load metadata for docker.io/library/caddy:2.5.2-alpine                                                                                                                                                                              1.2s
 => [internal] load metadata for docker.io/library/node:16                                                                                                                                                                                         1.2s
 => [stage-1 1/4] FROM docker.io/library/caddy:2.5.2-alpine@sha256:b31ff95e98737b849d6af1fb9d9cb54a66ba3684564b3310541f60b12b1dd619                                                                                                                1.4s
 => => resolve docker.io/library/caddy:2.5.2-alpine@sha256:b31ff95e98737b849d6af1fb9d9cb54a66ba3684564b3310541f60b12b1dd619                                                                                                                        0.0s
 => => sha256:fd0c7d01ba8a98bee02450f9f44e2eac5030b38a232f4af1d7c6e816d8230364 304.51kB / 304.51kB                                                                                                                                                 0.1s
....... a bunch of extract
 => => extracting sha256:513d779256048c961239af5f500589330546b072775217272e19ffae1635e98e                                                                                                                                                          7.9s
 => => extracting sha256:ae3b95bbaa61ce24cefdd89e7c74d6fbd7713b2bcae93af47063d06bd7e02172                                                                                                                                                          0.0s
 => => extracting sha256:0e421f66aff42bb069dffc26af6d132194b22a1082b08c5ef7cd69c627783c04                                                                                                                                                          1.5s
 => => extracting sha256:ca266fd6192108b67fb57b74753a8c4ca5d8bd458baae3d4df7ce9f42dedcc1d                                                                                                                                                          0.2s
 => => extracting sha256:ee7d78be1eb92caf6ae84fc3af736b23eca018d5dedc967ae5bdee6d7082403b                                                                                                                                                          0.0s
 => [internal] load build context                                                                                                                                                                                                                  0.5s
 => => transferring context: 49.02MB                                                                                                                                                                                                               0.4s
 => [internal] preparing inline document                                                                                                                                                                                                           0.0s
 => [stage-1 2/4] WORKDIR /src                                                                                                                                                                                                                     0.2s
 => [build 2/6] WORKDIR /src                                                                                                                                                                                                                       0.6s
 => [build 3/6] COPY . ./                                                                                                                                                                                                                          0.2s
 => [build 4/6] RUN corepack enable                                                                                                                                                                                                                0.5s
 => ERROR [build 5/6] RUN yarn install --immutable                                                                                                                                                                                                 1.6s
------
 > [build 5/6] RUN yarn install --immutable:
#0 1.546 Internal Error: Cannot find module '/src/.yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs'
#0 1.546 Require stack:
#0 1.546 - /root/.cache/node/corepack/yarn/3.6.3/yarn.js
#0 1.546 Require stack:
#0 1.546 - /root/.cache/node/corepack/yarn/3.6.3/yarn.js
#0 1.546     at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1028:15)
#0 1.546     at Function.Module._load (node:internal/modules/cjs/loader:873:27)
#0 1.546     at Module.require (node:internal/modules/cjs/loader:1100:19)
#0 1.546     at require2 (/usr/local/lib/node_modules/corepack/dist/corepack.js:41313:24)
#0 1.546     at u3 (/root/.cache/node/corepack/yarn/3.6.3/yarn.js:390:12580)
#0 1.546     at mf (/root/.cache/node/corepack/yarn/3.6.3/yarn.js:390:13423)
#0 1.546     at D (/root/.cache/node/corepack/yarn/3.6.3/yarn.js:391:21777)
#0 1.546     at Function.find (/root/.cache/node/corepack/yarn/3.6.3/yarn.js:391:22383)
#0 1.546     at async i (/root/.cache/node/corepack/yarn/3.6.3/yarn.js:446:10705)
#0 1.546     at async t (/root/.cache/node/corepack/yarn/3.6.3/yarn.js:446:10283)
------
Dockerfile:7
--------------------
   5 |     
   6 |     RUN corepack enable
   7 | >>> RUN yarn install --immutable
   8 |     
   9 |     RUN yarn run web:build:prod
--------------------
ERROR: failed to solve: process "/bin/sh -c yarn install --immutable" did not complete successfully: exit code: 1
```
- Version:
- Platform:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

1. Clone the latest main branch. 
2. Run `docker build . -t lichtblick`

**Expected Behavior**

The Docker image should be built.

ctw-joao-luis, 2025-01-15:
Hey @t0ny-peng thanks for reaching out! So I've tried to reproduce your error but I had no problems building the Docker image, and looking at the error you're showing us it looks like you haven't enabled corepack. So after cloning the repo you should:

1. $ git lfs pull

t0ny-peng, 2025-01-16:
Hi @ctw-joao-luis I've git lfs pull on my host. May I ask if `corepack` and the `yarn` command should run on the host or inside the Docker container?

luluiz, 2025-01-22:
Hi @t0ny-peng 
If you intend to run the application in Docker, you should run the commands inside the Docker container.
But we tested the Dockerfile and it works properly.
Please, clear your docker cache building with --no-cache flag.

closed 2025-01-22 by luluiz