lichtblick-suite/lichtblick #324: Cannot load layout

go-sakayori, 2025-01-15:
**Description**
I have saved my layout on lichtblick. When I try to import this layout to another PC nothing happens and the latest layout would be kept. The dialog box disappears once I have selected the file. 

- Version:1.8.0
- Platform: Ubuntu
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): NA

**Steps To Reproduce**
1. Export json file by View -> Export layout to file
2. Bring layout to another PC (or change layout on lichtblick by adding/deleting panels)
3. View -> Import layout from file -> Select the json file in the GUI.

**Expected Behavior**
The layout should be changed according to the json file.

ctw-joao-luis, 2025-01-15:
Hey @go-sakayori, so we've tried to reproduce your error and both import and export layout worked fine between on our devices.
I created a custom layout exported it and passed it to a colleague of mine and we had no problems. If you're having problems with exporting a layout after applying changes to it you need to make sure you save those changes before exporting the layout again (as you can see on the image below).

[image]

If if the issue still persists or it's something else unrelated to what I said above could you add a bit more context and the layout you're having problems with?

Thank you in advance, 
Lichtblick Team

go-sakayori, 2025-01-15:
@ctw-joao-luis Thank you for your quick response and support.
I had a misunderstanding with the layouts. I thought by clicking the icon -> View -> Import layout from file would work (call this `procedure icon`).
[image]
Now I found that the Layouts tab as you've showed in your image is suitable for applying any layouts.
Export seems to be working perfectly, but importing layouts seems to fail when importing from the `procedure icon`.
Does the above procedure work in your environment?

ctw-joao-luis, 2025-01-27:
Hey @go-sakayori sorry for taking so long to reply, but yeah answering to your question im not having any issues import or exporting whether from the layouts tab or via the menu reached through the icon.

closed 2025-01-29 by ctw-joao-luis