lichtblick-suite/lichtblick #330: Ubuntu 24.04: yarn desktop:start FAIL

jdsika, 2025-01-22:
**Description**

- Version: v1.9.0
- Platform: Ubuntu 24.04 LTS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**
Follow instructions to build.
Run `yarn desktop:start`

> 
> [85262:0122/133843.570671:FATAL:setuid_sandbox_host.cc(163)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /home/USER/lichtblick/node_modules/electron/dist/chrome-sandbox is owned by root and has mode 4755.
> /home/USER/lichtblick/node_modules/electron/dist/electron exited with signal SIGTRAP

**Solution**
Issue with electron. See [here](https://github.com/electron/electron/issues/42510#issuecomment-2171583086)

jdsika, 2025-01-22:
I actually cannot start it successfully:

> yarn desktop:start 
> [HMR] Waiting for update signal from WDS...
> Loaded language from settings: undefined
> Set language: en
> Lichtblick 1.9.0
> Could not set app as handler for lichtblick://
> Elapsed (ms) until new StudioWindow: 121
> APPIMAGE env is not defined, current application is not an AppImage
> New Lichtblick window 1
> Automatic updates disabled (development environment)
> Starting update loop
> Installing Chrome extensions for development...
> (node:87668) UnhandledPromiseRejectionWarning: Error: Invalid header: Does not start with Cr24
>     at crxToZip (webpack-internal:///../../node_modules/unzip-crx-3/dist/index.js:32:15)
>     at eval (webpack-internal:///../../node_modules/unzip-crx-3/dist/index.js:69:32)
>     at callHanler (webpack-internal:///../../node_modules/yaku/lib/yaku.js:719:16)
>     at tryCatcher (webpack-internal:///../../node_modules/yaku/lib/yaku.js:436:32)
>     at eval (webpack-internal:///../../node_modules/yaku/lib/yaku.js:575:38)
>     at flush (webpack-internal:///../../node_modules/yaku/lib/yaku.js:478:17)
>     at process.processTicksAndRejections (node:internal/process/task_queues:77:11)
> (Use `electron --trace-warnings ...` to show where the warning was created)
> (node:87668) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)

Blank screen appears:

[image]

laisspportugal, 2025-01-22:
Hello @jdsika, thank you for your feedback.

It's related to the bug mentioned on another issue and it will be solved by the PR 
#331

closed 2025-01-22 by laisspportugal

jdsika, 2025-01-22:
Uh, sorry did not see that!