lichtblick-suite/lichtblick #382: Excessive calls of converter function when registering two message converters for same message type

thomassedlmayer, 2025-02-20:
I created a minimal extension that just registers two message converters for the same message (to SceneUpdate and FrameTransform):

https://github.com/thomassedlmayer/lichtblick-extension-bug-demo

This results in excessive calls of the converter function for various frames (see timestamps in console logs) of the dataset when loading an mcap file (and setting the topic on visible!). This can be seen in the Developer Tool Console.

I added an example mcap file in the same demo repo.

jdsika, 2025-02-20:
@laisspportugal this is a major problem

aneuwald-ctw, 2025-04-01:
Hi @thomassedlmayer and @jdsika,

Thank you for reporting this issue and for providing a minimal extension to reproduce it. Our team has conducted an in-depth investigation, and we can confirm that message converters are being called correctly, even when multiple converters exist for the same data type.

However, we identified that Lichtblick handles message conversions through two different mechanisms:

1. Current Frame Processing
- When a panel subscribes to a topic, it retrieves the closest message to the current timestamp for conversion and rendering.
- Message converters are correctly executed in this scenario, though we see opportunities for optimization.

2. BlockLoader Preloading
- Some panels require full message history rather than just the current frame, triggering Lichtblick's BlockLoader to preload messages in chunks (each block covering ~1% of the dataset).
- In this process, messages converted to `foxglove.FrameTransform` are preloaded into memory alongside the original messages. Since your MCAP contains a single topic, this results in nearly double the memory usage (decompressed ~1GB). This is the primary cause of the BlockLoader reaching its memory limit.
- Additionally, we observed that converters are executed excessively (over **7,000 times in your case**) instead of just once per message. This unnecessary duplication impacts performance, and our team has prioritized fixing it.

### Next Steps
We appreciate you bringing this to our attention, as it helped us uncover inefficiencies in the conversion process. Our team will address the redundant conversions in the BlockLoader to enhance performance and memory management. We will update this issue as soon as improvements are available.

Thanks again for your contribution!

Best regards,
Alex & the Lichtblick Dev Team

aneuwald-ctw, 2025-04-01:
Adding to the last message, you can also deactivate the preloading messages if you want in the transforms settings:

[image]

thomassedlmayer, 2025-04-09:
@aneuwald-ctw Thank you for taking the time to check out the issue and give that much information on the reasons/internals!

> Some panels require full message history rather than just the current frame, triggering Lichtblick's BlockLoader to preload messages in chunks (each block covering ~1% of the dataset).

Would it make sense (or even be possible) to implement something similar to H264 I-Frames containing a complete snapshot of the data every now and then so that you don't have to load from the beginning? Or is this what the block loading is actually doing?

Could it be a reasonable feature for extensions to be able to tell Lichtblick that your extension's data does not need to be preloaded? In our case every frame is completely self contained. As far as I understand, the preloading can only be turned on/off for the whole application.

@aneuwald-ctw Could you give some heads up as soon as the "over 7000 calls"-bug is fixed? I would then go ahead and test it with our extension. Currently, preloading NEEDS to be turned off before loading our extension files so that Lichtblick does not lag like crazy or crash after some minutes (at least that's what happens on my laptop even for quite small files).

aneuwald-ctw, 2025-04-10:
> Would it make sense (or even be possible) to implement something similar to H264 I-Frames containing a complete snapshot of the data every now and then so that you don't have to load from the beginning? Or is this what the block loading is actually doing?

@thomassedlmayer, I don't believe that would be a good ideia. Video compression is perfect for visualization but you sacrifice some quality, and that will not be done in Lichtblick. And when I say "frame" I don't mean pixels like a frame of a video, but "frame" on Lichtblick context is all the data that will be visualized in that moment. So no, we can't do like key frames and do some interpolation between them, Lichtblick will lose a lot of details and users would not be able to analyze correctly their data.

> Could it be a reasonable feature for extensions to be able to tell Lichtblick that your extension's data does not need to be preloaded? In our case every frame is completely self contained. As far as I understand, the preloading can only be turned on/off for the whole application.

About this, data needs to be preloaded for different reasons, it gets very difficult to make one single point to turn on and off. The user needs to understand how it works and decide when it will be a good idea to turn off.

> [@aneuwald-ctw](https://github.com/aneuwald-ctw) Could you give some heads up as soon as the "over 7000 calls"-bug is fixed? I would then go ahead and test it with our extension. Currently, [...]

Absolutely, I will let you know when we fix that!

thomassedlmayer, 2025-04-10:
> Video compression is perfect for visualization but you sacrifice some quality, and that will not be done in Lichtblick.

I think we ran into some major misunderstanding here ;) I am aware of the different meanings of the word "frame". I actually talked about using "something like" the concept of intra-frames (complete independent state snapshots) in order to prevent the need to load everything from the beginning. I'm not talking about using video compression.
As far as I understood so far, Lichtblick sometimes needs to preload all data from the beginning for accumulating frame transforms for example. If you had certain frames (snapshots) in certain time intervals that represent a self-contained/independent state of a frame transform, you wouldn't have to go all the way back to the start in order to find out the current position. You would only have to go back to the last snapshot.

I hope, you get what I meant. I just wanted to share/discuss some ideas that seemed like sensible concepts/solutions for the drawback of preloading that seem to exist for certain situations (especially large files in combination with extensions).

Let's see how it works as soon as the apparent bug is solved!

aneuwald-ctw, 2025-04-22:
Yes @thomassedlmayer, I understand the difference. The video compression was an example where this "intra-frames" approach works very well, but I strongly believe that would not be a great solution for Lichtblick.

However, once Lichtblick an open source project, feel free to implement it by yourself and try out this approach. If you can reach the point where you can have this snapshots working without losing data and improving this performance, you can open then a PR and we for sure can analyze together.

But for now, we will focus on finding out why the block loader is triggering so many times the message converters. When we fix this, we will close this issue.

jdsika, 2025-05-05:
> Adding to the last message, you can also deactivate the preloading messages if you want in the transforms settings:
> <img alt="Image" width="412" src="https://private-user-images.githubusercontent.com/162589552/429059074-7a074fab-e390-4ec8-acf6-da4e5c1dfb10.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDY0MzUyMzUsIm5iZiI6MTc0NjQzNDkzNSwicGF0aCI6Ii8xNjI1ODk1NTIvNDI5MDU5MDc0LTdhMDc0ZmFiLWUzOTAtNGVjOC1hY2Y2LWRhNGU1YzFkZmIxMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDUwNVQwODQ4NTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZTdkZTAyMjExZmQyNTU3NTE3YjM5ZDIwMDMyN2ZmZjQ2OTU2ZTBmMmEwMjgxNTA0MWJkNGFjNzVjNjNlNjYyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.WGJmp89eBrXfHsJ8EIzmuo_cH5aC-zNAhO4TI4HJ1MQ">

Dear @aneuwald-ctw . I have opened this suggestion to make sure that users do not run into the pre-loading bug: https://github.com/lichtblick-suite/lichtblick/discussions/491
I received some complaints that every time were resolved by telling them about the disabling of the pre-loading.

aneuwald-ctw, 2025-05-19:
Hello @jdsika,

I’ve replied to you in the discussion: https://github.com/lichtblick-suite/lichtblick/discussions/491#discussioncomment-13147486

We are considering making this change, but we’d like to understand whether you’ve considered the potential loss of value for the use cases that users have been reporting to you. Of course, turning the feature off will improve performance, since fewer messages need to be processed and no additional computation is required. But this feature exists for a reason.

Are you confident that disabling it by default won’t negatively impact the quality or usefulness of the tool in those scenarios?

aneuwald-ctw, 2025-05-19:
Another thing is, by the MCAP you have shown us, your use case use all the data from 1 or 2 topics. That for sure is bad for preloading messages because it does not preload only the part of the message you use, but it saves all the message in memory. Having all your messages in a few topics can be problematic indeed. Did you try / it is possible to split your data in more topics? At least the TF messages? that way you would preload only those TF messages and the performance for sure would improve a lot, without this "quick workaround" of disabling the preload tf

aneuwald-ctw, 2025-05-19:
Yes, I understand. It is a more specific use case. Unfortunately Lichtlbick is made for a broad use cases in robotics, not specific for Open Simulation Interface standard. That's why we are a little concerned in make a change that would improve your use case, but can decrease quality in more general and common use cases, that we expect to be most of the community.

About "disabling preload single topics", Lichtblick only preloads a topic when this message has something to be preloaded. Or because it was subscribed in an extension, or if it's a TF (and TF Preload is on) or even for charts etc. We can try to make something to disable by default and save in user configuration (that way, you have to disable once), but the default will keep being preloading on. 

Also, I'm afraid that you will keep having problems, as when you try to create new charts or something like that, and we will keep needing to "adjust" Lichtblick for your specific use case.

One more thing Is, we are studying new ways to improve the block loader performance. It is not something to be released soon but we are trying our best to make it works more performant.

thomassedlmayer, 2025-05-19:
> Unfortunately Lichtlbick is made for a broad use cases in robotics, not specific for Open Simulation Interface standard.

But this issue is not really specific to Open Simulation Interface. You could see it as general improvement of supporting large data topics that might not require preloading. Don't these occur in other robotic/sensoric use cases as well (just wondering)? E.g. when working with original sensor interfaces and mcap, especially when having large sensor captures.

> That's why we are a little concerned in make a change that would improve your use case, but can decrease quality in more general and common use cases, that we expect to be most of the community.

But that's why I proposed to keep preloading active by default but additionally offer some flag to indicate that any frame transforms (or whatever) coming from some specific topic do not need to be preloaded because they're self-contained and not accumulative.

> Also, I'm afraid that you will keep having problems, as when you try to create new charts or something like that, and we will keep needing to "adjust" Lichtblick for your specific use case.

So preloading is not just about frame transforms but also used for other features (charts as you said)? What are the things a topic must contain in order to be preloaded? Or what decides if a topic is preloaded or not? To me these things are not really clear yet. Is there some documentation we can check out in order to understand if we might run into problems if we decide to rely on deactivated preloading in the future?

thomassedlmayer, 2025-10-16:
Hi @aneuwald-ctw,

I've had a look at some internals of Lichtblick and tried to make my point again by showing an immature demo of what would make the OSI MCAP data visualization faster by orders of magnitude (>700ms down to stable <50ms updates):
https://github.com/lichtblick-suite/lichtblick/compare/develop...thomassedlmayer:lichtblick:feature/demo-self-contained-data-update

I identified that the addOrUpdateEntity function takes a really large portion of time for each render step for large frames and it's called many times in between render points (even up to 50 or 60 times for my example files). Updating the scene for every incoming frame even if it's not rendered by the next rendering point is really useless for use cases with strictly non-cumulative data (snapshot data). Lichtblick's full compliance with cumulative data for updating its scene is fine in general for potentially cumulative data. But if the same principle is applied to strict snapshot data (especially large datasets) this leads to unnecessary accumulation of step sizes because the longer it takes to process one frame, the more new scene updates arrive in the background and the more scene updates need to be performed before rendering the next frame. Instead, only the last received frame could be used to update the scene before rendering it right away. The large amount of unnecessary scene updates for large datasets leads to a buildup of scene update backlog which can only be fixed by pausing the visualization at some point. Once you wait a few seconds and press resume, it runs smoothly for some frames until the backlog builds up again really quickly and it starts to slow down more and more the longer you let it run.

Do you see the potential/possibility to have this kind of coalescing scene updating principle (or basically frame dropping) as an option for SceneUpdate data that comes with some kind of flag which guarantees that it is a self-contained snapshot and not cumulative?

I did not invest a lot of time into writing this code and didn't investigate the impact on other features but as far as I could tell, this worked surprisingly well for OSI MCAP files with large frames out of the box. Of course, it will give you wrong data for cumulative data right now (at least when frame drops occur). I am aware that this leads to outdated scene states in between rendering points, so it might require decoupling visualization from other stuff, if there is anything relying on the scene state. Also, relocating the addOrUpdateEntity from handleSceneUpdate to startFrame might have some impact somewhere. I just gave it a try and surprisingly didn't notice any issues so far.

Do you have any insight on potential pitfalls or why a differentiation of cumulative and snapshot data couldn't be an option? Would it be possible for you to investigate the feasibility?

@jdsika

jdsika, 2026-10-17:
@thomassedlmayer we need to check which of our issues are still relevant