lichtblick-suite/lichtblick #404: Extension: Readme and Changelog missing

jdsika, 2025-03-09:
Hi team,

my extension contains the README and CHANGLEOG in the .foxe archive but it is not displayed when loaded into Lichtblick:
https://github.com/Lichtblick-Suite/asam-osi-converter/releases/tag/v0.0.5

[image]

aneuwald-ctw, 2025-03-10:
Hey @jdsika ,

Thanks for bringing this to our attention! This isn't an issue specific to your extension but rather a general limitation affecting all extensions. We've taken note of it and will handle it internally. We'll work on a fix as soon as we can.

thomassedlmayer, 2025-03-12:
@aneuwald-ctw Would it be already enough to merge this PR: https://github.com/Lichtblick-Suite/extension-registry/pull/1 ?

Does Lichtblick pull the readme/changelog information from the Lichtblick's extension-registry or from foxglove's extension-registry?

aneuwald-ctw, 2025-04-21:
Hi @thomassedlmayer,

No, that repository is no longer in use and has been archived — it’s an old artifact from the original fork. Currently, Lichtblick only parses and displays data from the Marketplace / Extension Registry.

We’ve already opened an internal issue to address this behavior, and it will be resolved in the future.

Best regards,
Alex

PR #495 (2025-05-09): Adding readme and changelog to extension details
**User-Facing Changes**
Lichtblick is now able to show both readme and changelog information related to the extensions installed locally, just like it did to the extensions installed via the Marketplace

[image]

[image]

**Description**
There were two main problems that Lichtblick had with this feature for extensions installed locally:
- First the type of objects the `ExtensionDetails` component was receiving to list the local extensions didn't have the `readme` and `changelog` attirbutes;
- Second was that the function that was returning the v [... cut]
commit f64e8742 (2025-05-09): Adding readme and changelog to extension details (#495)

**User-Facing Changes**
Lichtblick is now able to show both readme and changelog information
related to the extensions installed locally, just like it did to the
extensions installed via the Marketplace

[image]

[image]

**Description**
There were two main problems that Lichtblick had with this feature for
extensions installed locally:
- First the type of objects the `ExtensionDetails` component was
receiving to list the local extensions didn't have the `readme` and
`changelog` attirbutes;
- Second was that the function that was returni [... cut]
closed 2025-05-09 by ctw-joao-luis