lichtblick-suite/lichtblick #405: Custom topic settings in extension

thomassedlmayer, 2025-03-12:
I tried to create custom panel settings for the 3D panel in a Lichtblick extension. I want to use settings to switch visualization options for the extension's scene entities. I thought it should be possible to use the `event` (of type `MessageEvent`) object to read the topic's config (`event.topicConfig`).

Type definition of registerMessageConverter arguments:
```
export type RegisterMessageConverterArgs<Src> = {
  fromSchemaName: string;
  toSchemaName: string;
  converter: (msg: Src, event: Immutable<MessageEvent<Src>>) => unknown;
  /**
   * Custom settings for the topics using the schema specified in the *toSchemaName* property
   */
  panelSettings?: Record<string, PanelSettings<unknown>>;
};
```

```
const generatePanelSettings = <T>(obj: PanelSettings<T>) => obj as PanelSettings<unknown>;

enum DataTypes {
  OPTION1,
  OPTION2,
}

type Config = { shape: DataTypes };

export function activate(extensionContext: ExtensionContext): void {

  extensionContext.registerMessageConverter({
    fromSchemaName: "osi3.SensorView",
    toSchemaName: "foxglove.SceneUpdate",
    converter: (osiSensorView: SensorView, event: Immutable<MessageEvent<SensorView>>) => {
      console.log(event);
      return {
        deletions: [],
        entities: [],
      };
    },
    panelSettings: {
      "3D": generatePanelSettings({
        settings: (config) => ({
          fields: {
            shape: {
              label: "Options",
              input: "select",
              options: [
                { label: "OPTION1", value: DataTypes.OPTION1 },
                { label: "OPTION2", value: DataTypes.OPTION2 },
              ],
              value: config?.shape,
            },
          },
        }),
        handler: (action, config: Config | undefined) => {
          console.log(action);
          console.log("handler");
          if (config == undefined) {
            return;
          }
          if (action.action === "update" && action.payload.path[2] === "shape") {
            config.shape = action.payload.value as DataTypes;
          }
        },
        defaultConfig: {
          shape: DataTypes.OPTION1,
        },
      }),
    },
  });
}
```

[image]

The issue is that the MessageEvent object (argument of the converter function) contains a topicConfig that is undefined (see next image). Shouldn't it also contain the other options (`Color`, `Show outlines` and `Selection variable`)?

[image]

Also, the handler function is apparently not being called. At least I'm not getting any console logs when switching between the options.

Did I do something wrong? Unfortunately, I don't find any examples or documentation on how to define custom panel settings. I basically just reverse engineered most of the code based on the type definitions.

aneuwald-ctw, 2025-03-14:
Hello @thomassedlmayer, thanks for reaching out!

Regarding the `topicConfig`, this issue was fixed in version 1.10 of Lichtblick. Could you confirm that you're using the latest version?

As for the `handler` function not being called, we are already aware of this issue and will address it as soon as possible. We appreciate your patience and will keep you updated on any progress.

Please let us know which version of Lichtblick you are currently using!

thomassedlmayer, 2025-03-14:
@aneuwald-ctw Ah I see. I forgot to update my debugging build from 1.9 to 1.10. I can confirm that I see the default topic options now. Still, the handler is not called. Good to hear that you're already working on it. Thank you for letting me know!

thomassedlmayer, 2025-03-14:
Wait, I just noticed that the event.topicConfig also contains my custom setting. I just thought it was missing because there were only three settings:

[image]

One of them is actually my custom setting (shape). The "select variable" setting is not supposed to appear there? So the handler call is just supposed to immediately trigger the config update?

aneuwald-ctw, 2025-03-14:
> [@aneuwald-ctw](https://github.com/aneuwald-ctw) Ah I see. I forgot to update my debugging build from 1.9 to 1.10. I can confirm that I see the default topic options now. Still, the handler is not called. Good to hear that you're already working on it. Thank you for letting me know!

Glad that it worked :)

aneuwald-ctw, 2025-03-14:
> Wait, I just noticed that the event.topicConfig also contains my custom setting. I just thought it was missing because there were only three settings:
>
> [image]
>
> One of them is actually my custom [...]

No no, the selection variable is a special case and it is added to the global variables in Lichtblick. It is not accessible in message converters, but it is accessible via extensionContext, subscribing to `variables`:

[image]

If you see in the right panel, it was created a variable called `my_id_test`, which I defined in the `selection variable` field:

[image]

If you share more informations about what you want to achieve, maybe our team can help you to find a solution!

aneuwald-ctw, 2025-03-15:
@thomassedlmayer just correcting myself, this property DOES appear in the `topicConfig`, but only if it is defined. As default it is undefined so it is not sent. I enter a new id string in the input, and it started to appear in `topicConfig` in the message converter as `selectedIdVariable`:

[image]

Hope that solves your questions. If it does, I kindly ask you to close this issue.

Thanks again your contribution!

closed 2025-03-17 by thomassedlmayer