lichtblick-suite/lichtblick #409: Files not loading from timestamp 0.0 when opening

thomassedlmayer, 2025-03-14:
I noticed that when I open files with Lichtblick, the initially loaded frame is not first frame of the file but some later point in time. (I just open the file and don't click anything; the white line indicates the current position in the file; the yellow line indicates "timestamp 0")

[image]
(0.09899997711s into the file)

[image]
(0.099s into the file)

I'm currently on Lichtblick 1.10.0. This happens with various mcap files containing different types of topics.

Is there some reason behind this and could this be fixed so that Lichtblick loads the first frame of the file when opening it?

This is a problem because we try to automatically log some content out of a Lichtblick extension and having to manually jump back to the first frame of the file invalidates the order of the frames in the log file.

aneuwald-ctw, 2025-03-14:
Hello @thomassedlmayer, thanks for reaching out!

Yes, this behavior is expected in Lichtblick. It has been present since the project was forked from Foxglove, and if you check there, they still use the same approach.

The reason behind this is that the timeline advances by one step (100ms) when opening a file to ensure that the first messages are properly read. Lichtblick does not look for messages at an exact timestamp but instead finds the last available message before the current timestamp across all topics. This initial step prevents the screen from starting completely black and ensures that some content is available for rendering.

However, this does not mean that any messages are lost. For example, I created an MCAP file with a log topic running at 100Hz, and in the first render, all 10 log messages (covering the first 100ms) are already read. This confirms that the messages are still processed correctly:

[image]

That said, I understand how this could be problematic for automated logging in your extension. We can consider adding an option to start at the exact first timestamp instead of advancing by 100ms. Or even you can use `allFrames` instead of `currentFrame`, that would guarantee that the data is loaded from the beginning. Would that work for your use case?

Let us know your thoughts!

thomassedlmayer, 2025-03-14:
@aneuwald-ctw Thank you a lot for the quick reply and your explanation!

This is some weird "intended" behaviour though. It probably shouldn't advance some random time, right? If you have a file with very fine granular time steps (e.g. 0.001s), this would just just iterate through lots of frames without the user being able to do anything about it. To be honest, I don't exactly get why this workaround is necessary in the first place. I thought the timeline would start with the first frame of the file, doesn't it? So if I have a file with its first message time of 100.0s (log_time? publish_time?), the timeline should start at 100.0s and Lichtblick should load the first frame when I open the file. Or which time does the timeline even display?

One question regarding your log messages in the screenshot: Did you somehow write these log messages as you traversed the file or were these log messages contained in the file before you opened it in Lichtblick?

But as far as I understand it now, Lichtblick basically just traverses through the frames up to 0.099 in the background and then pauses. So if I load a file and press "play", the order of the extension hook calls should be consecutive starting from the first frame? I will check if this works.

> Or even you can use allFrames instead of currentFrame, that would guarantee that the data is loaded from the beginning. Would that work for your use case?

I don't exactly understand what `allFrames` or `currentFrame` means. Loading all the data from the beginning sounds problematic to me (see pre-loading issue with extensions). The perfect feature for our "automated logging problem" would be an option to run Lichtblick headless and immediately start to traverse the file frame by frame from the first to the last. We just want to reliably extract (currently done via REST API calls and a logging server) the scene update stuff from the extension. "Reliably" means preventing duplicate frames in the logs through potential interactions with the GUI.

aneuwald-ctw, 2025-04-10:
@thomassedlmayer Apologies for the delayed response.

You're right that in your specific case—where all data is published on a single topic—the first message may contain everything needed for rendering the initial frame. However, this setup is neither typical nor ideal in most scenarios.

Lichtblick is designed to support a wide range of use cases, especially those involving multiple topics with messages that are not perfectly synchronized. To ensure a meaningful first render, the player performs an initial buffering step to collect a sufficient set of messages from different topics. This avoids displaying an empty or incomplete screen at the very beginning. For example, if one topic (e.g., GPS) only publishes every 10 seconds, it's expected that some early frames may not include that data. 

This behavior, starting slightly ahead of the first message, is intentional and has proven to work well in standard workflows. That said, we understand this might not align with your specific setup.

If you need full control over the initial playback time, you can implement a custom extension and programmatically call `seekPlayback` from the `PanelExtensionContext`. This will allow you to seek to the exact time you want when your panel is initialized.

Let us know if this approach works for your use case!

thomassedlmayer, 2025-04-10:
> @thomassedlmayer Apologies for the delayed response.

No worries!

I'm really not just talking about our OSI extension point of view by the way. I don't get why it would be preferable to start at some random timestamp that is not even related to actual message intervals in a specific file.

Let's have a look at the great example you brought up:

> For example, if one topic (e.g., GPS) only publishes every 10 seconds, it's expected that some early frames may not include that data.

1. GPS message is published at -1.0s
2. At 0.0s the data capturing starts
3. Next GPS message is published at +9.0s (first GPS message contained in the file)

Loading 0.099s of the file doesn't really solve the issue for the GPS data, right? The GPS data should not (and never! even if you scroll to 9s and go back to 0) be displayed until timestamp 9.0.
Why would you favor messages with a publish time that is lower than 0.099s? Why would you call the frame at 0.099 a "meaningful render"? My opinion: If at 0.0 there are no messages in certain topics, then show the user that there are no messages and render it like that! Let them scroll to 0.099 if they think that 0.099 is their first meaningful render.

In my understanding Lichtblick should be a "true" data viewer and not a viewer that fools the user by hiding that there is a lack of data in certain initial time intervals. For me as a user, I would like to know that there is no data in a certain topic until a certain frame. This actually should be clearly displayed somewhere.

I hope I could make you somehow understand where I'm coming from. My concern is not about our data not fitting into that concept. I also worry that this approach might be unknowingly irritating people working with multiple topics with varying message intervals. I don't understand the issue with showing the user a black screen or a lack of data in the initial frame. Is this a technical issue or is it for the sake of users that are irritated by a black/incomplete screen when loading a file?

> If you need full control over the initial playback time, you can implement a custom extension and programmatically call seekPlayback from the PanelExtensionContext. This will allow you to seek to the exact time you want when your panel is initialized.

Thanks for the info, didn't know this is possible. I will have a look! Is it planned to have some kind of documentation or at least feature overview for extensions? Or is there something already which I have not seen yet?

aneuwald-ctw, 2025-04-15:
Hi @thomassedlmayer, thanks again for your thoughtful input.

We understand your perspective regarding the playback start time, and it's clear you've put a lot of thought into the implications for users dealing with sparse or irregular message intervals. Our team has considered this behavior, and while we recognize that there might be edge cases where an earlier start time could make sense, at the moment we don't see it as a priority for change. We haven't received feedback from other users indicating this is a widespread issue, so for now, we're not planning any modifications in this area.

That said, we're glad the `seekPlayback` approach might help you achieve your desired behavior within your extension. If you run into issues implementing it, feel free to reach out.

Regarding documentation, we agree that better extension support and examples would be valuable, and this is something we're working on improving over time.

Since there are no further actions planned from our side, we’ll close this issue for now. If you’d like to continue the discussion or gather community feedback, we encourage you to open a topic under the [Discussions tab](https://github.com/lichtblick-suite/lichtblick/discussions).

Best regards,
Alex

closed 2025-04-15 by aneuwald-ctw