lichtblick-suite/lichtblick #426: Cannot load ROS1 bags

kira-oli, 2025-03-26:
**Description**
Lichblick cannot load ros1 bag files.

- Version: 1.11.0, 1.10.0
- Platform: Ubuntu 20.04.6 LTS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS1 bag file

**Steps To Reproduce**
- open ROS1 bag file
- no data source displayed

**Expected Behavior**
- ROS1 messages available

ctw-joao-luis, 2025-03-26:
Hey @kira-oli, thanks for reaching out! 
We’ve just tested it on our end and it seems that `.bag` files can’t be opened through the `Open local files...` option. However, if you drag a `.bag` file directly into Lichtblick, it will open and work as expected. For now, please use this as a workaround, if possible, while we investigate the issue. We’ll let you know once we have more information! 
If you run into any more issues in the meantime, feel free to reach out!

Best regards,
LB Team.

kira-oli, 2025-03-27:
Hey @ctw-joao-luis! 

Thanks for the fast feedback! 
I can definitely work with that.

Best regards,
Oli

ctw-joao-luis, 2025-04-04:
Hi @kira-oli!
Just an update, this issue has been fixed and merged (#447) into our main branch. It will be included in the next release, coming soon. 
Thanks for your patience!

closed 2025-04-04 by ctw-joao-luis