lichtblick-suite/lichtblick #449: No info box content updates

thomassedlmayer, 2025-04-10:
**Description**
I noticed that the object-specific info box data is not updated when playing a file. You need to click the object again for it to update. Is this supposed to be like this?

- Version: 1.10.5, 1.11.0
- Platform: Windows 10
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
  - https://github.com/thomassedlmayer/lichtblick-extension-bug-demo/blob/master/20250110T150000Z_sv_3.7.0_highway_drop-bike_5ms.mcap
  - Extension: https://github.com/lichtblick-suite/asam-osi-converter/releases/tag/v0.0.5

**Steps To Reproduce**
[image]

**Expected Behavior**
I would have expected that the info box always shows the current state of the object when it's selected.

ctw-joao-luis, 2025-04-10:
Hey @thomassedlmayer, thanks for reporting this!

I was able to reproduce the issue described and we're going to investigate it, I'll let you know when there's more updates.

Best regards, 
Lichtblick Team.

ctw-joao-luis, 2025-04-21:
Hey again @thomassedlmayer! 
So we've looked into this and discovered that the `Selected Object` modal was never built  to have the data it shows automatically updated as the file is read. We think he could be useful and we're going to study if  implementing such feature is feasible, but for now we suggest the use of the Raw Message panel if you really need to see your data updating on a specific object. 
You can even use the little pop up icon on the top right of the modal that opens a Raw Message panel with the selected object's data 

[image]

Thanks for continually reporting issues and suggestions to help Lichtblick grow! I'm going to close this issue, but feel to open a new one or a suggestion on our Discussions if you have any more questions/doubts!

Best regards, 
Joao Luis

closed 2025-04-21 by ctw-joao-luis

jdsika, 2025-04-24:
I would suggest to keep this issue as a reminder of this potential feature request?

ctw-joao-luis, 2025-04-28:
> I would suggest to keep this issue as a reminder of this potential feature request?

Hi @jdsika! As I mentioned earlier, we'll definitely keep this feature in mind for future implementation. For now, it's not a priority, but we’ll continue tracking it internally.

Best regards,
Joao