lichtblick-suite/lichtblick #461: LFS Quota Exceeded - Cannot Pull Required Files

hemkum, 2025-04-18:
**Description**

Unable to pull Git LFS files due to storage quota being exceeded.  
This prevents cloning and building the project properly. Are these lfs files present in the released source code ?

- **Version:** v1.11.0 (also tried with v1.10.0)
- **Platform:** macOS
- **Data source:** N/A (Problem occurs during repository setup itself)

**Steps To Reproduce**

1. Clone the repository: 
2. Navigate into the project folder:
3. git lfs pull

response: This repository exceeded its LFS budget. The account responsible for the budget should increase it to restore access. Failed to fetch some objects from 'https://github.com/lichtblick-suite/lichtblick.git/info/lfs'

**Expected Behavior**
- `git lfs pull` should complete successfully without quota errors.
- All LFS-tracked files (e.g., plugins `.cjs` files, image thumbnails) should be downloaded properly so that the project can be built and used.

montanier, 2025-04-19:
I have this error as well

SiHao24, 2025-04-21:
I have this error as well， and

SiHao24, 2025-04-21:
I have this error as well
Platform: Windows 10
node version: v23.9.0

when i clone repo, The error is as follows：

Downloading .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs (1.1 MB)
Error downloading object: .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs (af9e38a): Smudge error: Error downloading .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs (af9e38a38ec93526f6b6053b45c657bbb7bcbc76f05edf0c5c910d9f7309a76e): batch response: This repositor
y exceeded its LFS budget. The account responsible for the budget should increase it to restore access.

and I copy files from  .yarn/plugins/@yarnpkg/, add them to the folder,

`git lfs fetch --all`

The errors is as follows:
batch response: This repository exceeded its LFS budget. The account responsible for the budget should increase it to restore access.
batch response: This repository exceeded its LFS budget. The account responsible for the budget should increase it to restore access.
error: failed to fetch some objects from 'https://github.com/lichtblick-suite/lichtblick.git/info/lfs'

ctw-joao-luis, 2025-04-21:
Hey all, thanks for reaching out. We've also noticed this issue and we're working to find a way to solve it. Sorry for any inconvenience and when there's an update I'll let you know!

Best regards,
Joao Luis

aneuwald-ctw, 2025-04-22:
Hello,

We have found a solution, you can check it out in -> #466 #465 

Basically we removed the dependency of Git LFS from our project.

Any questions, please don't hesitate to reach out. Please continue this discussion on #466 

Best regards,
Alexandre.

closed 2025-04-22 by aneuwald-ctw