lichtblick-suite/lichtblick #493: Opening mcap file via launch argument does not work anymore

thomassedlmayer, 2025-05-05:
**Description**
My Lichtblick installation just automatically updated from 1.11 to 1.12. Now I can't open mcap files (mcap containing OSI messages used for our asam-osi-converter) by double-clicking them in Windows Exporer anymore. The same applies when starting a file by running `start Lichtblick.exe <path-to-mcap-file>`. Lichtblick opens but the windows stays grey. The window title quickly switches between "Lichtblick" and the filename. Nothing more happens when I wait some minutes. It also happens to me if I don't have any extension installed.

When Lichtblick is already open and I try to open a file from the explorer or cmd I get the following error before it completely crashes:
[image]

In order to open files I now have to open Lichtblick and select the file via Lichtblick's GUI.

- Version: 1.12
- Platform: Windows
- Data source: https://github.com/thomassedlmayer/lichtblick-extension-bug-demo/blob/master/20250110T150000Z_sv_3.7.0_highway_drop-bike_5ms.mcap

**Steps To Reproduce**
Download the mcap and try to double-click it or run `start Lichtblick.exe <path-to-mcap-file>`.

closed 2025-05-06 by gabriela-almeida-ctw

jdsika, 2025-05-07:
I would kindly suggest to let the one close the issue who opend it as this person needs to confirm that the issue was resolved properly and in addition maybe close the issue when it was included in a release as other users do not know if the issue is known, addressed/resolved because you normally do not check closed issues after encountering a problem in the current release

carolinabamorim, 2025-05-07:
Hi @jdsika, that's a good point, however, the current workflow of closing issues when a PR is merged ensures that we're staying efficient and not holding up progress waiting for a release. The issue might not always be resolved immediately in a given release, and closing it based on a PR link helps signal that the issue has been addressed at least in code.
As for waiting until it’s included in a release—users generally don’t expect an issue to remain open indefinitely while waiting for a release to be deployed. Closing the issue when it's resolved in code is a common practice in many open-source projects, and it allows us to keep the process moving smoothly.
That said, since this particular issue is crucial to user experience, we are actively working on a release that will include the fix. I understand that a release can be an important milestone, but the intent is to make sure the process stays clear and efficient for everyone. 
Hope that clears things up!