lichtblick-suite/lichtblick #503: Settings API setting changes are not being applied to current frame.

MeganOosthoekKodiak, 2025-05-19:
**Description**
1. The custom settings are not being applied to the currentFrame
2. The config UI is laggy to update

https://github.com/user-attachments/assets/97b87832-fb1a-48ce-993f-39465b10b151

- Version: 1.13.4
- Platform: linux
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): mcap

**Steps To Reproduce**

i created a small extension: 
[helloworld.zip](https://github.com/user-attachments/files/20312156/helloworld.zip)

mcap: 
[pose_steps.zip](https://github.com/user-attachments/files/20312291/pose_steps.zip)

**Expected Behavior**
config ui is not laggy + custom setting gets applied to current frame

aneuwald-ctw, 2025-05-20:
Hello @MeganOosthoekKodiak, thanks for reaching us!

Yes, it is a known limitation we have right now. About the laggy behavior we are still investigating how to fix it.

About not being applied to the current Frame, that is a little bit different, because the Settings API area applied to the message converter, and a message converter is triggered only when you have new frames. Make Settings API re-trigger after a change is some delicate. We are planning on act on this, but can't say for sure that it will be possible to achieve this soon.

We will return to you on this issue as soon as we have some solutions.

Cheers

jdsika, 2025-06-24:
The missing re-trigger of the converter plugins at the current frame after the settings change is a behavior that confuses our users as well. We just explain that you have to press play again but a fix will definitely improve user experience.