lichtblick-suite/lichtblick #514: Testing large files (over 1GB) in Lichtblick, the system becomes unresponsive or crashes.

JaplaVikash, 2025-05-27:
**Description**
When testing large files (over 1GB) in Lichtblick, the system becomes unresponsive or crashes. Is there a solution for testing large datasets in Lichtblick without causing system issues?

Environment Details
Platform: [e.g., Ubuntu 22.04, Windows 10, etc.]
Data Source: [e.g., .bag file, .mcap file, rosbridge, ROS 1/2 native.

Steps to Reproduce
Open Lichtblick.
Load a data file larger than 1GB.
Observe the system behavior (e.g., crash, freeze, high memory usage).

**Expected Behavior**

Lichtblick should be able to handle large files (over 1GB) without crashing. The system should either load the data successfully or provide a fallback mechanism or warning if the file is too large.

ctw-joao-luis, 2025-05-27:
Hey @JaplaVikash,

First of all, thanks for reaching out and for using Lichtblick!
Regarding the issue you're experiencing: Lichtblick does have some known limitations when handling large files, primarily due to the 4GB heap memory cap imposed by the V8 engine it's built on. However, this limitation isn't directly tied to file size alone, we’ve seen cases where multi-gigabyte files (up to 50Gbs) work just fine.
The more common problem tends to be the structure of the file itself. For example, files with a very large number of messages or a high message frequency can cause performance issues, including out-of-memory crashes. 

Could you share the message list from your file? You can use the [MCAP CLI tool](https://mcap.dev/guides/cli) to extract that information. Also if you could share with us more info about your environment that could help us trying to understand what's going wrong. For instance are you using a lot of extensions? How's the layout you use, does it have many 3D panels or Raw Message panels? Could you even share your files with us so we can try to reproduce the problem? 

Best regards, 
Joao Luis

JaplaVikash, 2025-06-03:
> Hey [@JaplaVikash](https://github.com/JaplaVikash),
>
> First of all, thanks for reaching out and for using Lichtblick! Regarding the issue you're experiencing: Lichtblick does have some known [...]

JaplaVikash, 2025-06-03:
Yes, my file is a 25GB .mcap file, and we want to visualize multiple extensions at once. Can you please explain how to use the MCAP CLI?

closed 2025-06-03 by JaplaVikash