lichtblick-suite/lichtblick #571: How and where can  increase the memory and  chase allocation in Lichtblick? Because when I open the panel, it crashes.

JaplaVikash, 2025-07-03:
**Description**
when I open the multi panel, it crashes.

- Version: latest release 
- Platform:window , ubuntu
- Data source (e.g. bag file, mcap file, ):

**Steps To Reproduce**

**Expected Behavior**

JaplaVikash, 2025-07-09:
[

https://github.com/user-attachments/assets/bed455bf-66a3-4311-91cb-05d6aea44a3c
https://github.com/user-attachments/assets/78be87ef-8ff6-4246-8637-3532fa13de72

](url)

In above  video file visualizes data correctly in Foxglove, but in Lichtblick, it shows buffering. Please help me understand where I can increase heap memory or improve performance to reduce buffering and enable faster data visualization.

JaplaVikash, 2025-07-10:
Thanks for the reply. I will check the same file. Sorry, I had sent the wrong video earlier. Please check this one—it's working fine in Foxglove.

https://github.com/user-attachments/assets/95417ce4-b3f1-469f-9ad9-22d287399739

JaplaVikash, 2025-07-10:
Thanks for the reply. I will check the same file. Sorry, I had sent the wrong video earlier. Please check this one—it's working fine in Foxglove.[

https://github.com/user-attachments/assets/5e7f6be3-6711-4925-8543-f19c426295ca

https://github.com/user-attachments/assets/71969acd-2fb4-4fb9-9735-fd754d8fd854

](url)

aneuwald-ctw, 2025-07-10:
Ok thanks for the reply! We will keep investigating to make Lichtblick more performant.

In any case, if you want to join on this as well, be welcome!

JaplaVikash, 2025-07-10:
Thanks and sure. Yes, I’m also investigating.

JaplaVikash, 2025-07-16:
HI All,
I noticed that the Foxglove team added support for buffering messages from MCAP sources as raw Uint8Arrays in commit #7376.(https://github.com/lichtblick-suite/lichtblick/commit/41e3f14508f02921b9575ab4129345a817544bc7)
I just wanted to confirm—has your team also integrated this feature into Lichtblick?

aneuwald-ctw, 2025-07-17:
> HI All, I noticed that the Foxglove team added support for buffering messages from MCAP sources as raw Uint8Arrays in commit #7376.([41e3f14](https://github.com/lichtblick-suite/lichtblick/commit/41e3f14508f02921b9575ab4129345a817544bc7)) I just wanted to confirm—has your team also integrated this feature into Lichtblick?

Hey @JaplaVikash thanks for mentioning that. Yes, we are currently working on this, you can check PR #586, it will be available very soon!

Cheers,
Alex

JaplaVikash, 2025-07-18:
Thanks a lot @aneuwald-ctw

closed 2025-07-21 by carolinabamorim