lichtblick-suite/lichtblick #580: Bug?

jdsika, 2025-07-11:
**Description**
When loading an .mcap file the loading bar is always jumping to a time frame >0s.
 
[image]

**Steps To Reproduce**
Use v1.17.0 and load an .mcap file.

**Expected Behavior**
The **first** message in the .mcap file is displayed.

aneuwald-ctw, 2025-07-14:
Just to add, I know that in your case of simulation, it is a single topic that holds all the information, so the "first message" makes sense. But in most scenarios, there are multiple topics with different frequencies. So the question is: how to determine which timestamp should Lichtblick navigate? That's why there is this "guess" of 99ms.

jdsika, 2025-07-21:
Could you explain why it is not possible to grep the "first message" without guessing which time stamp this message could be? It is not necessarily the case that the first message in simulation is at time = 0 but can be something different - it is rather that I would wish to display the first message of the .mcap file that I am loading.

carolinabamorim, 2025-07-22:
Hi @jdsika how the mcap is read is by date and time, so we know which is the first message. 
However, in your specific case you have one single topic. We are generic enough to users who have for example 100 topics and subscribe to 20, which topic should Lichtblick get for the first message? It can be tricky here. I understand that in your specific case you would like things differently and we will work on a solution to make it flexible enough 🤗