lichtblick-suite/lichtblick #592: The 3D view of the mcap file cannot be rendered normally

BrainLeeNJIT, 2025-07-17:
I open a mcap file,  The Image is ok, but the 3D view is  not correct,   use this mcap file， I open it on https://open.platform.io-ai.tech/,  its  3D view  is no  problem

aneuwald-ctw, 2025-07-17:
Hello @BrainLeeNJIT, thanks for reaching out.

We would need more information to understand the problem, for instance:
- The MCAP you are using (it is not in the original issue)
- How this platform works? (it is not maintained by us)
- Do you need extensions?
- What is wrong in the 3d? the frame transformation? entities are not rendered?
- Which version of Lichtblick are you using?
- Which OS are you using?

Please provide as much information as possible so we can better assist you.

Cheers,
Alexandre

carolinabamorim, 2025-07-21:
@BrainLeeNJIT if you still need our assistance please reopen the issue with all the information required.