lichtblick-suite/lichtblick #602: Desktop version does not run successfully

dhuruvasaditya, 2025-07-29:
**Description**
`yarn desktop:start` pops up a blank white screen with an html error. GUI does not render succesfully.

- Version: node v22.17.0
- Platform: Ubuntu 22.04.5 LTS

**Steps To Reproduce**

```
athurvas@bos-robotics-l01:~/Desktop/new/lichtblick$ corepack enable
athurvas@bos-robotics-l01:~/Desktop/new/lichtblick$ yarn install
.....
➤ YN0000: ┌ Fetch step
➤ YN0013: │ yn@npm:3.1.1 can't be found in the cache and will be fetched from the remote registry
➤ YN0013: │ yocto-queue@npm:0.1.0 can't be found in the cache and will be fetched from the remote registry
➤ YN0013: │ yocto-queue@npm:1.2.1 can't be found in the cache and will be fetched from the remote registry
➤ YN0013: │ zustand@npm:4.5.7 can't be found in the cache and will be fetched from the remote registry
➤ YN0013: │ zwitch@npm:2.0.4 can't be found in the cache and will be fetched from the remote registry
➤ YN0000: └ Completed in 1s 155ms
➤ YN0000: ┌ Link step
➤ YN0007: │ electron@npm:37.1.0 must be built because it never has been before or the last one failed
➤ YN0007: │ esbuild@npm:0.25.0 must be built because it never has been before or the last one failed
➤ YN0007: │ protobufjs@patch:protobufjs@npm%3A7.2.5#../../patches/protobufjs.patch::version=7.2.5&hash=b342bd&locator=%40lichtblick%2Fmcap-support%40workspace%3Apackages%2Fmcap-support must be built because it never has been before or the last one failed
➤ YN0007: │ core-js-pure@npm:3.39.0 must be built because it never has been before or the last one failed
➤ YN0007: │ unrs-resolver@npm:1.10.1 must be built because it never has been before or the last one failed
➤ YN0000: └ Completed in 8s 271ms
➤ YN0000: Done with warnings in 9s 728ms

athurvas@bos-robotics-l01:~/Desktop/new/lichtblick$ yarn desktop:serve
....
quicklook:
  assets by chunk 2.37 MiB (id hint: vendors)
    asset vendors-node_modules_lichtblick_wasm-bz2_dist_index_js.js 1.56 MiB [emitted] (id hint: vendors)
    asset vendors-node_modules_lichtblick_wasm-zstd_dist_index_js.js 577 KiB [emitted] (id hint: vendors)
    asset vendors-node_modules_buffer_index_js.js 149 KiB [emitted] (id hint: vendors)
    asset vendors-node_modules_lichtblick_wasm-lz4_dist_index_js.js 60.7 KiB [emitted] (id hint: vendors)
    asset vendors-node_modules_path-browserify_index_js.js 42.8 KiB [emitted] (id hint: vendors)
  asset main.js 8.67 MiB [emitted] (name: main)
  asset _2da9.js 766 bytes [emitted]
  asset _7f03.js 766 bytes [emitted]
  asset _96a7.js 766 bytes [emitted]
  asset index.html 211 bytes [emitted]
  orphan modules 401 KiB [orphan] 325 modules
  runtime modules 32.5 KiB 17 modules
  javascript modules 3.16 MiB
    modules by path ../../node_modules/ 3.05 MiB 389 modules
    modules by path ../../packages/ 118 KiB 39 modules
    + 4 modules
  asset modules 2.15 MiB
    modules by path ../../node_modules/ 1.89 MiB
      ../../node_modules/@lichtblick/wasm-zstd/dist/wasm-zstd.wasm 540 KiB [built] [code generated]
      + 2 modules
    modules by path ../../packages/suite-desktop/ 262 KiB
      ../../packages/suite-desktop/resources/icon/BagIcon.png 130 KiB [built] [code generated]
      ../../packages/suite-desktop/resources/icon/McapIcon.png 132 KiB [built] [code generated]
  ../../node_modules/protobufjs/google/protobuf/descriptor.json 7.04 KiB [built] [code generated]
  quicklook (webpack 5.99.9) compiled successfully in 7947 ms
No errors found.

```

The error:

```
$ yarn desktop:start
[HMR] Waiting for update signal from WDS...
Loaded language from settings: undefined
Set language: en
Lichtblick 1.18.0
Could not set app as handler for lichtblick://
Elapsed (ms) until new StudioWindow: 58
APPIMAGE env is not defined, current application is not an AppImage
New Lichtblick window 1
Automatic updates disabled (development environment)
Starting update loop
Installing Chrome extensions for development...
(electron) 'session.getAllExtensions' is deprecated and will be removed. Please use 'session.extensions.getAllExtensions' instead.
(electron) 'session.loadExtension' is deprecated and will be removed. Please use 'session.extensions.loadExtension' instead.
Finished extension install: {
  id: 'epndoelgkappodbihlhmjkghahdokjla',
  name: 'React Developer Tools',
  path: '/home/athurvas/.config/Lichtblick Dev/extensions/fmkadmapgofadopljbjfkapdkoienihi',
  url: 'chrome-extension://epndoelgkappodbihlhmjkghahdokjla/',
  version: '6.1.5',
  manifest: {
    action: { default_icon: [Object], default_popup: 'popups/disabled.html' },
    background: { service_worker: 'build/background.js' },
    content_scripts: [ [Object] ],
    content_security_policy: { extension_pages: "script-src 'self'; object-src 'self'" },
    description: 'Adds React debugging tools to the Chrome Developer Tools.\n' +
      '\n' +
      'Created from revision 5d87cd2244 on 7/4/2025.',
    devtools_page: 'main.html',
    host_permissions: [ '<all_urls>' ],
    icons: {
      '16': 'icons/16-production.png',
      '32': 'icons/32-production.png',
      '48': 'icons/48-production.png',
      '128': 'icons/128-production.png'
    },
    manifest_version: 3,
    minimum_chrome_version: '114',
    name: 'React Developer Tools',
    optional_permissions: [ 'clipboardWrite' ],
    permissions: [ 'scripting', 'storage', 'tabs' ],
    update_url: 'https://clients2.google.com/service/update2/crx',
    version: '6.1.5',
    version_name: '6.1.5 (7/4/2025)',
    web_accessible_resources: [ [Object] ]
  }
}
window.loadURL(http://localhost:8080/renderer/index.html)
load-pending-files
injectFilesToOpen: no files - skipping
window URL loaded
[172822:0729/184958.200272:ERROR:CONSOLE:1] "Request Autofill.enable failed. {"code":-32601,"message":"'Autofill.enable' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[172822:0729/184958.200299:ERROR:CONSOLE:1] "Request Autofill.setAddresses failed. {"code":-32601,"message":"'Autofill.setAddresses' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[172822:0729/184958.267976:ERROR:CONSOLE:2] "Electron sandboxed_renderer.bundle.js script failed to run", source: node:electron/js2c/sandbox_bundle (2)
[172822:0729/184958.268041:ERROR:CONSOLE:2] "TypeError: object null is not iterable (cannot read property Symbol(Symbol.iterator))", source: node:electron/js2c/sandbox_bundle (2)

```

A White screen loads up with debug console which says 
```
index.js:594 [webpack-dev-server] Server started: Hot Module Replacement enabled, Live Reloading enabled, Progress disabled, Overlay enabled.
log.js:39 [HMR] Waiting for update signal from WDS...
index.ts:42 Lichtblick 1.19.0
index.ts:43 initializing preloader, argv="/home/athurvas/Desktop/lichtblick/node_modules/electron/dist/electron --type=renderer --enable-crash-reporter=4fc10648-db11-4b6d-837b-3ab3fd780af7,no_channel --user-data-dir=/home/athurvas/.config/Lichtblick Dev --fetch-schemes=package --app-path=/home/athurvas/Desktop/lichtblick/desktop/.webpack --no-sandbox --no-zygote --enable-experimental-web-platform-features --lang=en-US --num-raster-threads=4 --enable-main-frame-before-activation --renderer-client-id=4 --time-ticks-at-unix-epoch=-1753823283128004 --launch-time-ticks=5588740659 --shared-files=v8_context_snapshot_data:100 --field-trial-handle=3,i,5641814083498984163,15630273088051569474,262144 --enable-features=PdfUseShowSaveFilePicker --disable-features=ScreenAIOCREnabled,SpareRendererForSitePerProcess --variations-seed-version --allowCrashReporting=1 --allowTelemetry=1 --deepLinks=W10="
index.html:1  Failed to load resource: the server responded with a status of 404 (Not Found)
```

aneuwald-ctw, 2025-07-30:
Hey @dhuruvasaditya, can you please share the logs after the desktop:serve? You put like 3 dots in that part.

athurvas@bos-robotics-l01:~/Desktop/new/lichtblick$ yarn desktop:serve
....
quicklook:

Try to guarantee that your desktop:serve is running in localhost:8080. What happens is, if the port is busy, it will try the 8081 and Electron will always try the localhost:8080

dhuruvasaditya, 2025-07-31:
> Hey [@dhuruvasaditya](https://github.com/dhuruvasaditya), can you please share the logs after the desktop:serve? You put like 3 dots in that part. [...]

Thank you for your response. Yes sorry the log was too long and seemed everything had run properly hence I cut it short. 
And yes it was a port problem it works fine now. Another application of mine was interfering with it, I did not think it through

closed 2025-07-31 by dhuruvasaditya