lichtblick-suite/lichtblick #604: Cannot  connect to Lichtblick using the Foxglove WebSocket method.

lubidkdh, 2025-07-31:
Dear Lichtblick Team
I can connect to Foxglove using the Foxglove WebSocket method, but I cannot use the Foxglove WebSocket method to connect to Lichtblick.
System win11
Lichtblick version 1.19.0
foxglove-sdk version 0.10.1 

This is the error message from my Lichtblick：

[image]

This is my code：
```
import foxglove
import time

from foxglove.schemas import Log, LogLevel, Timestamp

server = foxglove.start_server(host="0.0.0.0", port=8765)

while True:
    foxglove.log(
        "/hello",
        Log(
            timestamp=Timestamp.now(),
            level=LogLevel.Info,
            message="Hello, Foxglove!",
        )
    )

    time.sleep(0.033)

```

[image]

aneuwald, 2025-08-04:
Thanks for reaching out @lubidkdh. We will give a look on that and come back to you soon.

Best,
Alex

lubidkdh, 2025-08-05:
Thank you for your response！
I used the [@foxglove/ws-protocol](https://www.npmjs.com/package/@foxglove/ws-protocol) to stream live data to Lichtblick.But there were still some problems.
This is my bash command:
`npx @foxglove/ws-protocol-examples@latest mcap-play  D:\work\datasource\mcap\out.mcap`
This is the error message from my Lichtblick and Foxglove：

[image]

[image]

[image]
Btw， I wanted to visualize a large number of mcap files, but I only wanted to visualize the parts that I was interested in. So, at the beginning, I thought I could visualize the parts with specific timestamps in this way. However, disappointingly, I found that this method could only specify the playback speed for visualization and whether to control the loop playback.
The only solution I can think of for now is to use the mcap CLI to split the mcap file and then visualize it. However, this is very time-consuming and consumes a lot of storage space. Could you have any good suggestions or ideas?I'm really looking forward to your reply. Thanks!//

closed 2025-08-12 by lubidkdh

aneuwald-ctw, 2025-08-12:
Hey @lubidkdh have you found a solution?

lubidkdh, 2025-08-12:
yeah，Today, I re-uploaded several mcap files and found that they all played normally. Maybe it was due to the performance issue of the computer at that time.
But I still haven't found a way to visualize the mcap file with a specific timestamp. We hope that Lichtblick can update this function as soon as possible. Thank you very much!

AndBondStyle, 2025-10-08:
@aneuwald which websocket bridge package you recommend using? It looks like foxglove_bridge ROS package was recently updated to v3.2.0 (see changelog [here](https://index.ros.org/p/foxglove_bridge/#jazzy) at the bottom) and it is no longer compatible with the old protocol. Do you plan to adapt the new SDK-based protocol, or fork the bridge package and stay on the old one?