lichtblick-suite/lichtblick #663: New extension version does not replace old version

pablopilotti, 2025-08-27:
**Description**
When installing a new version of an extension, the old version is not replaced. Instead, both versions remain active in parallel.

- Version: All
- Platform: All
- Data source : N/A

**Steps To Reproduce**
1. Install an extension by drag and drop.
2. Install a newer version of the same extension (also via drag and drop).
3. Observe that the old version is still present alongside the new one.

[image]

**Expected Behavior**
The new extension version should override (replace) the previous one, leaving only the latest version installed.

aneuwald-ctw, 2025-09-15:
Hey @siliconstudio-yunying-hu, we gave a look on this issue and indeed that is not the correct behavior. We will work on that. Just a little clarification, we believe the correct is not install the newest over the oldest, but the current over the previous. What I mean with that is, if you want to do a downgrade for some reason, you can replace let's say your 0.0.2 version with the 0.0.1. 

We will let you know when we finish this implementation. Thanks for the contribution!

luluiz, 2025-10-15:
Hi, we have addressed and solved this issue in this PR #709 
Thanks for the contribution!

closed 2025-10-15 by luluiz